leaphy-robotics/leaphy-client #28: Buzz block generates incorrect code

bastiaanb, 2021-01-08:
Adding a buzz block generates an incomplete '#include' line.

[image]

paulsiersma, 2021-01-08:
@bastiaanb I know it looks totally dodgy, but you will find that the code compiles and actually works. It's the result of the library that "cleans" the code before showing it in the code window. I think it interprets `<arduino.h>` as a codecomment to be cleaned up. Renaming this issue to reflect this.

bastiaanb, 2021-01-08:
I see. The other includes use " instead of <, so they don't get "cleaned".
In any case the code does *not* compile on my Linux system:
```
cmd: '/tmp/.org.chromium.Chromium.Basp6J compile --fqbn arduino:avr:uno /home/bbakker/.config/easybloqs/sketch/sketch.ino --build-path /home/bbakker/.config/easybloqs/sketch',
  stdout: '\n',
  stderr: '/home/bbakker/.config/easybloqs/sketch/sketch.ino:1:10: fatal error: arduino.h: No such file or directory\n' +
    ' #include <arduino.h>\n' +
    '          ^~~~~~~~~~~\n' +
    'compilation terminated.\n' +
    'Error during build: exit status 1\n'
```
I don't know whether this is Linux specific. Maybe good to try on a clean windows or mac too.

bastiaanb, 2021-01-08:
BTW, other sketches do compile and upload successfully.

paulsiersma, 2021-01-08:
I think it might indeed be something linux-specific that we have to figure out. The buzzer code was tested by Roeland, I _think_ also on a recently cleaned Windows machine. I propose we (you :)) open a separate issue for the compilation failure.

bastiaanb, 2021-01-08:
Found the issue already: the include should be `#include <Arduino.h>`. So I expect this to be an issue on Macs too....

paulsiersma, 2021-01-11:
> So I expect this to be an issue on Macs too....

Yes that could be, I doubt it's been tested on mac

closed 2021-01-13 by paulsiersma