leaphy-robotics/leaphy-client #54: Blockly - block-arms less high, text bigger fontsize

Roelandsmith, 2021-02-15:
Voor de leesbaarheid en de hoeveelheid blocklagen die je in je scherm kunt hebben, zou dit erg fijn zijn. Zeker ook voor het maken van werkboeken. (leesbaarheid en benodigde ruimte)

Roelandsmith, 2021-02-15:
- Filmpje van Blockly Developer Summit met een deep dive in de rendering pipeline
- Codelabs walkthrough om een custom renderer te maken en te gebruiken
- Walkthrough van een Blockly Renderer hackathon met referentie naar code

Roelandsmith, 2021-02-15:
- Filmpje van Blockly Developer Summit met een deep dive in de rendering pipeline: https://youtu.be/Er4JLJD1Ni8
- Codelabs walkthrough om een custom renderer te maken en te gebruiken: https://blocklycodelabs.dev/codelabs/custom-renderer/index.html?index=..%2F..index#0
- - Walkthrough van een Blockly Renderer hackathon met referentie naar code: https://sites.google.com/view/blockly-user-summit-2019/hackathon/rendering-quick-start

paulsiersma, 2021-02-16:
Moved to https://github.com/leaphy-robotics/leaphy-blockly/issues/57 and https://github.com/leaphy-robotics/leaphy-blockly/issues/58

closed 2021-02-16 by paulsiersma