leaphy-robotics/leaphy-client #98: Driver installer windows - update

Roelandsmith, 2021-04-08:
The usb-driver-installer-concept we use now appears to be confusing for the users. They neglect the pop-up whichs announces the need of installation. And when they do start installation, they get scared of the untrustworthy looks of the dialogue box and/or they get confused because the second box has no 'close'-button. (It keeps saying 'install') 

**Solutions:**

**1.  The driver-pop-up-box:** needs to get activated directly after the main installation. We hope users will still be in the 'istallation-mode themselves) (Instead of the situation now: it waits untill the first upload-session.)

**2. Proposal: we start using the driver that is also used by the mBlock-software:**
It looks a bit more 'trustworthy':
https://drive.google.com/file/d/1B96fPoJabtbiTXsIv-qkMiD5KvjYOsKx/view?usp=sharing

**3. The introduction-dialogue box needs more reassuring content:**
_Option 1 - if we succeed in using the mblock-version of the driver (see link above)_
We zijn bijna klaar! 
Je krijgt nu nog twee pop ups om de usb-driver te installeren.

[Oké, dat ga ik doen!] 

_Option 2 - (if we keep on using the driver used now in Easybloqs)_

We zijn bijna klaar! 
Je krijgt nog twee pop ups:
Bij pop-up 1 klik je op ‘OK’
Bij pop-up 2 klik je op ‘Install’.
Als het klaar is, druk je de pop-up weg met het kruisje.

[Oké, dat ga ik doen!]

paulsiersma, 2021-04-16:
Consider using the installer executable shared in Slack:
- It looks more trustworthy
- It closes automagically after finishing

paulsiersma, 2021-06-11:
Fixed with #113 , closing

closed 2021-06-11 by paulsiersma