leaphy-robotics/leaphy-client #108: RGB-sensor version 1.06 doesn't work

Roelandsmith, 2021-04-21:
It worked fine in version 1.03, but in version 1.06 not anymore.
We changed two things:
- RGB-blocks with drop down-feauture.
- The code in the 'read' block has been combined with the value-blocks. 

I think the second part is a problem. Jasper Simonis created the original blocks and I remember he did quite some research...so maybe it's a thing that is needed: a command block starting the RGB-sensor each time again.

Hope it can be fixed!
Back to the old system is fine you know (saves us work for redoing the lesson-levels too;)

paulsiersma, 2021-04-22:
@Roelandsmith What would really help is having the code of the same program as generated by 1.0.3 and as generated by 1.0.6. I cannot see how this code change would have caused this problem.

paulsiersma, 2021-04-26:
I think I have figured it out. Some of the code that moved to setup needs to be in the block code, so the sensor can be read more than one time. We can keep the updated block setup. PR coming soon

paulsiersma, 2021-04-27:
Moved to Blockly repo, will be fixed with #109

closed 2021-04-27 by paulsiersma