leaphy-robotics/leaphy-client #144: M1 MacBook support / Robot niet gevonden

Quinten0508, 2021-09-09:
Heb een Hyperdrive usb-c naar usb-a hub en sluit mijn Leaphy Original via daar op mijn M1 MacBook Air aan. EasyBloqs geeft aan de adruino drivers succesvol te hebben geinstalleerd, maar na op "uploaden naar robot" te klikken wordt er gemeld dat te robot niet kan worden gevonden. Mogelijk is dit omdat Easybloqs via Rosetta werkt en misschien daarom niet direct bij de robot kan, maar dat is slechts een gok.

paulleaphy, 2022-01-28:
Hoi @Quinten0508 ik heb helaas geen manier om dit te reproduceren dus oplossen wordt ook erg lastig. Werkt het wel als je de robot direct op je macbook aansluit? Dat zou ik dan willen aanraden als voor de hand liggende workaround. In de tussentijd sluit ik deze issue.

closed 2022-01-28 by paulleaphy

Quinten0508, 2022-01-28:
Het probleem ligt niet bij jullie, maar bij Arduino: https://github.com/arduino/arduino-create-agent/wiki
Het is voorlopig dus afwachten ben in bang. De issue kan inderdaad gesloten worden.