leaphy-robotics/leaphy-client #158: Vastloper in dialogbox

Roelandsmith, 2021-12-03:
Deze dook deze week op bij mij (Roeland) en ik kreeg hem ook doorgestuurd van een collega-docent.
De oplossing nu was: afsluiten en weer opstarten:)
[image]

paulsiersma, 2021-12-03:
@Roelandsmith zou je wat meer details kunnen geven wat het probleem precies is? 

Wat deed je? Wat verwachte je? Wat kreeg je daarvoor in de plaats? Gebeurt het elke keer of alleen onder bepaalde omstandigheden?

Dit maakt het voor mij veel makkelijker om het te reproduceren en te fixen.

Roelandsmith, 2021-12-03:
Hoi,

 

 

Hier weet ik helaas nog heel weinig van…

Het gebeurde en hij was ook helemaal vastgelopen.

Ik voeg de LOG die ik nu vond even toe….daar zou de gebeurtenis in moeten staan…

 

 

Groetjes,

 

Roeland

 

[2021-12-01 13:12:01.933] [info]  
 0 / 450463    0.00%
Updating index: package_index.json downloaded

 0 / 543    0.00%
Updating index: package_index.json.sig downloaded

[2021-12-01 13:12:03.383] [info]  
 0 / ?    0.00%
Updating index: library_index.json 0 B / ?    0.00%
Updating index: library_index.json 11.12 MiB / ?  116635.35%
Updating index: library_index.json downloaded

[2021-12-01 13:12:03.795] [info]  Required Core arduino:avr already installed, attempting upgrade...
[2021-12-01 13:12:04.223] [info]  Platform arduino:avr is already at the latest version

[2021-12-01 13:12:04.655] [info]  Required Library Leaphy Extra Extension not found, installing...
[2021-12-01 13:12:04.663] [info]  Required Library Servo already installed, attempting upgrade...
[2021-12-01 13:12:05.639] [info]  Leaphy Extra Extension depends on Leaphy Extra ***@***.***
Downloading Leaphy Extra ***@***.***
Leaphy Extra ***@***.*** already downloaded
Installing Leaphy Extra ***@***.***
Installed Leaphy Extra ***@***.***

[2021-12-01 13:12:05.648] [info]  Servo depends on ***@***.***
Downloading ***@***.***
Installing ***@***.***
Already installed ***@***.***

[2021-12-01 13:27:59.468] [info]  
 0 / 450463    0.00%
Updating index: package_index.json downloaded

 0 / 543    0.00%
Updating index: package_index.json.sig downloaded

[2021-12-01 13:29:13.916] [info]  
 0 / 450463    0.00%
Updating index: package_index.json downloaded

 0 / 543    0.00%
Updating index: package_index.json.sig downloaded

[2021-12-01 14:32:47.486] [info]  
 0 / 450463    0.00%
Updating index: package_index.json downloaded
[... 161 lines cut ...]
Updating index: library_index.json 14.83 MiB / ?  155542.51%
Updating index: library_index.json downloaded

[2021-12-01 17:19:23.412] [info]  Required Core arduino:avr already installed, attempting upgrade...
[2021-12-01 17:19:23.813] [info]  Platform arduino:avr is already at the latest version

[2021-12-01 17:19:24.178] [info]  Required Library Leaphy Extra Extension not found, installing...
[2021-12-01 17:19:24.184] [info]  Required Library Servo already installed, attempting upgrade...
[2021-12-01 17:19:25.052] [info]  Servo depends on ***@***.***
Downloading ***@***.***
Installing ***@***.***
Already installed ***@***.***

[2021-12-01 17:19:25.055] [info]  Leaphy Extra Extension depends on Leaphy Extra ***@***.***
Downloading Leaphy Extra ***@***.***
Leaphy Extra ***@***.*** already downloaded
Installing Leaphy Extra ***@***.***
Installed Leaphy Extra ***@***.***

[2021-12-01 17:19:48.745] [info]  
 0 / 450463    0.00%
Updating index: package_index.json downloaded

 0 / 543    0.00%
Updating index: package_index.json.sig downloaded

paulsiersma, 2021-12-03:
Hmm lastig te zeggen wat er precies mis is gegaan. Het lijkt er op alsof hij de robot niet kan vinden tijdens het uploaden. Erg lastig om zo op te pakken. Als het opnieuw optreedt moeten we goed proberen vast te leggen wat de omstandigheden waren en welke stappen leidden tot het optreden van de fout.

paulsiersma, 2021-12-29:
@Roelandsmith ik ga deze sluiten omdat het niet verder voorgekomen lijkt te zijn. Als het probleem toch nog speelt en je nieuwe informatie hebt kan je deze issue heropenen.

closed 2021-12-29 by paulsiersma