leaphy-robotics/leaphy-client #171: Code-Editor-button in startscherm

Roelandsmith, 2022-01-26:
Onder de robot-knoppen 'direct naar de editor'
knop maken.

Vraag: welke libraries laadt hij dan in?
Antwoord: die zijn als het goed is al ingeladen, want je hebt waarschijnlijk eerder met de blokken je robot geprogrammeerd
Vraag: maar als iemand Easybloqs installeert en direct de eerste keer naar de code-editor gaat...dan is er geen enkele library geinstalleerd(?)
Antwoord: voor de zekerheid bij keuze voor 'editor' alle mogelijke libraries ophalen? (In de praktijk zal dat snel gaan, want de meeste libraries zullen al ingeladen zijn bij eerder gebruik)

paulleaphy, 2022-01-28:
Straight to Code Editor, implementation idea notes:
1. Button selects the generic robot type
2. You are taken straight to the code editor view
3. Toggling to Blockly view is disabled

Roelandsmith, 2022-01-29:
Great!

A button in start screen would help a lot for user-friendliness. I think:)
And we need a warning: when you edited the code in the editor and you click back to 'block-screen' you lose all the edits you made in the code. So a lttle warning:  'Do you want to save the changes you made?'

And maybe a dialogue box: like an instruction / verwachtingsmanagement.
Blocks become code, but it doesnt work the other way around 
Blokken worden code, maar het werkt niet andersom.

And a dialoguebox:
When a program is saved in editor-mode, it beomes a special editor-code-file.

Als een programma opgeslagen wordt in de editor, wordt het een speciaal editor-bestand.

Roelandsmith, 2022-01-29:
nog wel open...

commit ce4241eb (2022-02-04): Fix #171 Allow going straight to code editor
closed 2022-02-04 by paulleaphy