leaphy-robotics/leaphy-client #176: compilatie error

DFromBalen, 2022-01-28:
You  get a compilation error 
when:
first:
i wanted to empty the Arduino on the leaphy robot.
So i uploaded the empty start  leayphy block 
That is not a good idea. 
When you  then want to upload a program  in  the Arduino you get a compilation error.
Restart Easybloqs and everything is just fine again .  no more compilation error 
so don't upload an empty start block

Second:
The other compilation error I get when i first load a program controlling the LED or the distance, sensor .
 the library   #include  "leaphyoriginal.h" is needed 
upload the program no problem. 

Then when you write or load a new program and you use use the Leaphy extra blocks to  par example  :control the ledstrip 
the library #include "ledstrip.h" is needed . When uploading to the Arduino you get a compiling error .
When You see the log file you  see a fatal upload error 

The second compiling error  also shows up when you do it vica versa  : First the ledstrip program  uploading then the led control program..
compiling error.
When restart  easybloqs problem partially solved you can upload  programs and that work whiteout compiling error.
As long as you keep in the same library. Need another library restart easybloqs 

is there a sollution ?

Dirk

paulleaphy, 2022-01-28:
@DFromBalen I'm sorry you are getting compilation errors. I'm able to reproduce this and will try to find a fix asap.

paulleaphy, 2022-01-28:
@DFromBalen I fixed the issue, it should not occur anymore in https://github.com/leaphy-robotics/leaphy-client/releases/tag/v1.1.1-pre.1 can you confirm?

DFromBalen, 2022-01-28:
Paul

 

No more compilation error 

We get the message uploading (code uploaden)

And uploading completed (upload voltooid)

 

Something is happening in the Arduino  (the rgb led is out for a few seconds)

But the old program starts running  

 

But the program is not uploaded in the Arduino of the Leafpy 

The old program stays on working 

 

Kind regards 

 

Dirk

paulleaphy, 2022-01-29:
Hi @DFromBalen, Dirk,

Thanks for testing this out. I think I got all the problems fixed in the [latest prerelease](https://github.com/leaphy-robotics/leaphy-client/releases/tag/v1.1.1-pre.2) (still building at the time of writing this). Please have a look if it works on your end. 

Kind regards, Paul

DFromBalen, 2022-01-29:
Paul 

 Very good news

everything works just fine now

No more compiling errors

An the uploading is also pico bello 

Switching projects and programs that  uses  different library’s  working just fine now

 

No more need for restarting easybloqs

No more issues 

I installed the    easybloqs-1.1.1-pre.3 now 

Thanks 

If you need a tester in the future ..let me know … 

I’m going to use the Leaphy in my coderdojo 

I just designed a 3D object to fix the led strip without  using screws 

Kind regards 

Met vriendelijke groeten 

Dirk

paulsiersma, 2022-01-31:
Hi @DFromBalen, Dirk. Good to hear. Thanks for your reporting the bug! If you spot any more, please let us know!

P.S. I've informed the hardware guru's about your 3D printed ledstrip holder. They may be in touch!

commit 4d9fb0e8 (2022-01-28): Merge pull request #179 from leaphy-robotics/compilation-bug

Fix #176 Get rid of build path in compile command
commit afdc1511 (2022-01-28): Fix #176 Get rid of build path in compile command
closed 2022-01-31 by paulsiersma