leaphy-robotics/leaphy-client #210: Error during build: exec: "/bin/avr-g++": file does not exist

Roelandsmith, 2022-03-17:
[2022-03-09 14:30:06.809] [info]  

[2022-03-16 13:28:29.975] [info]  

0 / 472675    0.00%

Updating index: package_index.json 0 B / 461.60 KiB    0.00%

Updating index: package_index.json 134.70 KiB / 461.60 KiB   29.18% 00m01s

Updating index: package_index.json 211.00 KiB / 461.60 KiB   45.71%

Updating index: package_index.json downloaded

 

0 / 543    0.00%

Updating index: package_index.json.sig downloaded

 

[2022-03-16 13:28:33.366] [info]  

0 / ?    0.00%

Updating index: library_index.json.gz 0 B / ?    0.00%

Updating index: library_index.json.gz 96.00 KiB / ?  983.04%

Updating index: library_index.json.gz 256.00 KiB / ?  2621.44%

Updating index: library_index.json.gz 512.00 KiB / ?  5242.88%

Updating index: library_index.json.gz 768.00 KiB / ?  7864.32% 00m01s

Updating index: library_index.json.gz 768.00 KiB / ?  7864.32% 00m01s

Updating index: library_index.json.gz 1.12 MiB / ?  11796.48% 00m01s

Updating index: library_index.json.gz 1.19 MiB / ?  12451.84% 00m01s

Updating index: library_index.json.gz 1.69 MiB / ?  17694.72% 00m01s

Updating index: library_index.json.gz 2.03 MiB / ?  21299.20% 00m02s

Updating index: library_index.json.gz downloaded

 

0 / 543    0.00%

Updating index: library_index.json.sig downloaded

 

[2022-03-16 13:28:34.314] [info]  Required Core arduino:avr not found, installing...

[2022-03-16 13:28:35.352] [info]  Platform arduino:avr@1.8.5 already installed

 

[2022-03-16 13:28:36.053] [info]  Required Library Leaphy Extra Extension not found, installing...

[2022-03-16 13:28:36.075] [info]  Required Library Servo not found, installing...

[2022-03-16 13:28:36.119] [info]  Required Library Adafruit GFX Library not found, installing...

[2022-03-16 13:28:36.159] [info]  Required Library Adafruit SSD1306 not found, installing...

[2022-03-16 13:28:36.202] [info]  Required Library Adafruit LSM9DS1 Library not found, installing...

[2022-03-16 13:28:36.248] [info]  Required Library Adafruit Unified Sensor not found, installing...

[2022-03-16 13:28:38.207] [info]  Downloading Servo@1.1.8...

Servo@1.1.8 already downloaded

Installing Servo@1.1.8...

Installed Servo@1.1.8

 

[2022-03-16 13:28:38.258] [info]  Downloading Adafruit GFX Library@1.10.13...

Adafruit GFX Library@1.10.13 already downloaded

Installing Adafruit GFX Library@1.10.13...

Installed Adafruit GFX Library@1.10.13

Downloading Adafruit BusIO@1.11.2...

Adafruit BusIO@1.11.2 already downloaded

Installing Adafruit BusIO@1.11.2...

Installed Adafruit BusIO@1.11.2

 

[2022-03-16 13:28:38.586] [info]  Downloading Adafruit Unified Sensor@1.1.5...

Adafruit Unified Sensor@1.1.5 already downloaded

Installing Adafruit Unified Sensor@1.1.5...

Installed Adafruit Unified Sensor@1.1.5

Downloading Adafruit LIS3MDL@1.1.0...

Adafruit LIS3MDL@1.1.0 already downloaded

Installing Adafruit LIS3MDL@1.1.0...

Installed Adafruit LIS3MDL@1.1.0

Downloading Adafruit BusIO@1.11.2...

Adafruit BusIO@1.11.2 already downloaded

Installing Adafruit BusIO@1.11.2...

Installed Adafruit BusIO@1.11.2

Downloading Adafruit LSM9DS1 Library@2.1.1...

 

0 / 14740    0.00%

Adafruit LSM9DS1 Library@2.1.1 downloaded

Installing Adafruit LSM9DS1 Library@2.1.1...

Installed Adafruit LSM9DS1 Library@2.1.1

 

[2022-03-16 13:29:15.668] [error] Compilation failed! Error: Command failed: C:\Users\ekare\AppData\Local\Temp\92cecd1d-245a-4c65-b25c-f6deb1c513e8.tmp.exe compile C:\Users\ekare\AppData\Roaming\easybloqs\sketch --fqbn arduino:avr:uno --build-path C:\Users\ekare\AppData\Roaming\easybloqs\bin --config-file C:\Users\ekare\AppData\Roaming\easybloqs\arduino-cli.yaml

Error during build: exec: "/bin/avr-g++": file does not exist

 

   at ChildProcess.exithandler (child_process.js:317:12)

   at ChildProcess.emit (events.js:315:20)

   at maybeClose (internal/child_process.js:1048:16)

   at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)

koen1711, 2024-02-03:
can't reproduce

koen1711, 2024-02-03:
can't reproduce

closed 2024-02-03 by koen1711