leaphy-robotics/leaphy-client #214: OLEd-Disaply compilation error

Roelandsmith, 2022-05-13:
Dit gebeurde op meerdere laptops. Het verversen van de library's werkte helaas niet.

De library staat gewoon netjes in de Leaphy_Eaxtra-librarymap, maar toch geeft hij onderstaande melding:

[2022-05-13 13:44:43.822] [error] Compilation failed! Error: Command failed: C:\Users\smithr\AppData\Local\Temp\19e53a69-468c-4e55-a497-727676c99c4b.tmp.exe compile C:\Users\smithr\AppData\Roaming\easybloqs\sketch --fqbn arduino:avr:uno --build-path C:\Users\smithr\AppData\Roaming\easybloqs\bin --config-file C:\Users\smithr\AppData\Roaming\easybloqs\arduino-cli.yaml
C:\Users\smithr\AppData\Roaming\easybloqs\sketch\sketch.ino:1:10: fatal error: OLED_Display.h: No such file or directory
 #include "OLED_Display.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
Error during build: exit status 1

    at ChildProcess.exithandler (child_process.js:317:12)
    at ChildProcess.emit (events.js:315:20)
    at maybeClose (internal/child_process.js:1048:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)

paulleaphy, 2022-05-21:
@Roelandsmith after looking for a long time, I think I found the issue. There was a problem in the leaphy-extra-extensions library, the weird part is to know how it ever worked with that problem. Now need to wait for Arduino Library Manager to pick up the new version. Than I can test whether this indeed fixes the issue.

Roelandsmith, 2022-05-24:
Check!

 

 

It’s a strange world😊

paulleaphy, 2022-05-24:
Hi @Roelandsmith does your comment mean it's resolved for you? For me the problem seems to be gone.

koen1711, 2023-09-20:
Seems to be fixed.

closed 2023-09-20 by koen1711