leaphy-robotics/leaphy-client #224: bij 1.2.1 pre 5 - blokken worden niet actief

Roelandsmith, 2022-10-07:
[image]

paulsiersma, 2022-10-13:
Hi Roeland,

Thanks for testing, and submitting a bug! I am unfortunately unable to reproduce this behaviour on my machine. To properly fix this bug, some more information would be really helpful. Could you provide it using the template below? To accommodate non-Dutch speaking contributors, English is preferred.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

koen1711, 2023-09-20:
Closed due to inactivity.

closed 2023-09-20 by koen1711